Snapmaker/Snapmaker-Artisan-Firmware-Issues #2: Bug: Z-Offset Bugs in the Artisan

mmack, 2023-03-14:


mmack, 2023-03-18:
Some more testing for you: It does not matter if i do first the z-offset calibration and then the bed leveling, or the other way around. But watch the 2 attached videos from the same print. The z value does not change at all while printing the first layer...

https://user-images.githubusercontent.com/100767/226094740-3be60a1f-0809-4806-a9d7-24fd4b98cda1.mp4

https://user-images.githubusercontent.com/100767/226094744-15d8b0ab-3d6b-4539-9438-3d23caf578a7.mp4

ganymedes, 2023-03-19:
I can confirm this issue, same happens with my artisan…!

mmack, 2023-03-23:
https://github.com/MarlinFirmware/Marlin/issues/24337 Might be the actual problem. 
Would be nice to see someone from snapmaker looking at this problem.

mmack, 2023-03-23:
I did a Bed Leveling with 80 degrees. Took one of the points, moved the printer to that position and queried the z-offset via M1006:

```
Recv: probed z: 17.055
Recv: Probing No. 52
Recv: x: 118.547249, y: 240.061859
[...]

Send: G0 X118.547249 Y240.061859 Z10
Recv: ok
[...]

Send: M1006
Recv: Tool Head: 3DP
Recv: fdm fault state: 0x0
Recv: fdm status: 2
Recv: live_z_offset  [0] = 0.000, [1] = 0.000
Recv: z compensation [0] = 0.507, [1] = 0.490

[...]
Recv: X:118.547 Y:240.062 Z:10.000 A:0.000 B:0.000 E:0.000 Count X:12012 Y:19412 Z:10998 A:0 B:0
Recv: ok

```

mmack, 2023-03-23:
And here another point for the reference, should the compensation not be changed?
```
Recv: probed z: 16.870
Recv: Probing No. 74
Recv: x: 268.344238, y: 164.837128
[...]
Send: G0 X268.344238 Y164.837128 Z10
Recv: ok
[...]
Send: M1006
Recv: Tool Head: 3DP
Recv: fdm fault state: 0x0
Recv: fdm status: 2
Recv: live_z_offset  [0] = 0.000, [1] = 0.000
Recv: z compensation [0] = 0.507, [1] = 0.490
````

ganymedes, 2023-03-23:
@sumibin any update on this issue?

mmack, 2023-03-23:
So i do more live debugging here and note what i've found out. So the z compensation is the difference between the nozzle and the live_z_offset is the value from the artisan console if one adds manual z levels.
The question still remains: How to check if the bed leveling index is actually used at all...

mmack, 2023-03-24:
And another finding:

I did the 3x3 Auto Leveling:
```
Recv: compensated Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 +17.638 +17.234 +16.902
Recv:  1 +17.718 +17.397 +16.975
Recv:  2 +17.479 +17.204 +16.745
```

Then i manually raised the second point in the grid (it's the front midth of the bed) for 2mm with this command: 
`M421 I1 J0 Q2.00`

So the Bed nowlooks like this:
```
Recv: compensated Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 +17.638 +19.234 +16.902
Recv:  1 +17.718 +17.397 +16.975
Recv:  2 +17.479 +17.204 +16.745
```

The calculated matrix looks like this:
```
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6
Recv:  0 +17.63800 +18.31548 +18.99296 +19.23400 +18.74763 +17.82481 +16.90200
Recv:  1 +17.67648 +18.06412 +18.45175 +18.56078 +18.20545 +17.57150 +16.93756
Recv:  2 +17.71496 +17.81276 +17.91054 +17.88755 +17.66326 +17.31819 +16.97311
Recv:  3 +17.71800 +17.61474 +17.51148 +17.39700 +17.26382 +17.11941 +16.97500
Recv:  4 +17.66196 +17.50565 +17.34932 +17.21089 +17.10226 +17.01152 +16.92078
Recv:  5 +17.57048 +17.44990 +17.32931 +17.20744 +17.08345 +16.95817 +16.83289
Recv:  6 +17.47900 +17.39415 +17.30930 +17.20400 +17.06463 +16.90481 +16.74500
```` 

So what happens now if i print a line there: 
At the beginning of the line the nozzle nearly is scratching. In the middle it's lifting, but not 2mm and at the end of the line (most forward right point) it's nearly scratching again.
See this photo:
[image]

But also keep note the line next to it. It's the before-print extrusion line (or whatever it's called). There i could visibly see that it made a 2mm lift... the nozzle was away from the bed 2mm. On the photo that's the wobbled line. 

Any ideas on that?

mmack, 2023-03-24:
I did it with the right nozzle. Only the right nozzle can use the complete Bed. Can you post your Bed Leveling Data here?

ganymedes, 2023-03-24:
@mmack sorry it worked once, can't reproduce it... i give up, there is a bug in the firmware! Plz fix it snapmaker!

KarstenSchwab, 2023-03-26:
Hi . Same Problem on my artisan. The kalibration is Not work . And also the kalibration is only in the middle and Not Nest enough To the outdide
[image]

KarstenSchwab, 2023-03-26:
[image]
Her
[image]
e some
![Uploading A7803623-11C1-42E6-881F-7145254C6DA6.jpeg…](
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

KarstenSchwab, 2023-03-26:
After all that kalibration i do my x x Test . On second try i do left 0,1 UP and Right 0.2 up 
[image]

KarstenSchwab, 2023-03-26:
After all the kalibration do x y . Not ok . I do left 0.1 up Right 0.2 
[image]

Th199403, 2023-03-27:
i have the same problem!  have tried everything

Th199403, 2023-03-29:
I have the same issue and its been days and haven’t been able to resolve it. Can you please explain how you used a 0.2mm sheet to calibrate your Artisan?

I have logged a support ticket with Snapmaker; I am not at all pleased with their initial reply. Here is my reply, in the hopes someone in the forum can explain to me what I am doing wrong if any.

REPLY TO SNAPMAKER:

I am afraid the recommendation isn’t relevant as I have been using the in-built calibration method to try and get a good Z and X/Y axis result. When using the in-built calibration process the bed is heated to 70 degrees Celsius for the Bed-Levelling and 65 degrees Celsius for X /Y Calibration. I have run the calibration tests for hours at a time trying all manner of auto and manual calibration tests even manually offsetting to the axis to compensate but to no avail. So I am confident the bed has been appropriately heated.

For your reference I am running firmware 2.2.1.

The issue here isn’t the X/Y axis; yes it looks terrible but its because the Z axis is way off. What I have attempted to do over the last few days is:

METHOD 1

I have ran the 9 x 9 81 point bed-levelling calibration
Then ran the Z axis auto calibration.
Then the X/Y axis calibration ← When running this calibration test the dual print head moves to the bottom left of the build plate
to print the initial pre-test strips; the left print head (Snapmaker Black PLA) hardly prints any plastic at all; in fact you can hear the brass head dragging over the glass PEI plate!
The right print head (Snapmaker White Support PLA) begins to print its initial pre-test strip; it for the most part prints normally.
RESULT: The actual X/Y test is a failure; with the results appearing to show that the black PLA print head is running far too near the print bed. ← Well this is what I think, not speaking from experience though.
METHOD 2

I have ran the 9 x 9 81 point bed-levelling calibration
Then ran the Z axis Sensor calibration. ← Because the Z Axis Auto Calibration doesn’t seem to be working.
Using the supplied white calibration card; I step the heads down manually until both Right and Left heads are touching the card, can be pulled toward me but cannot be pushed back under the head it wrinkles.
I then run the Z axis Auto Calibration
Then the X/Y axis calibration ← When running this calibration test the dual print head moves to the bottom left of the build plate
to print the initial pre-test strips; the left print head (Snapmaker Black PLA) hardly prints any plastic at all; in fact you can hear the brass head dragging over the glass PEI plate!
The right print head (Snapmaker White Support PLA) begins to print its initial pre-test strip; it for the most part prints normally.
RESULT: Failure with the X/Y test looking no different to METHOD 1 and the print head dragging across the print bed
METHOD 3

I have ran the 9 x 9 81 point bed-levelling calibration
Then ran the Z axis Sensor calibration. ← Because the Z Axis Auto Calibration doesn’t seem to be working.
Using the supplied white calibration card; I step the heads down manually until both Right and Left heads are touching the card, can be pulled toward me but cannot be pushed back under the head it wrinkles.
I then run the Z axis Auto Calibration
Then the X/Y axis calibration ← When running this calibration test the dual print head moves to the bottom left of the build plate
to print the initial pre-test strips; the left print head (Snapmaker Black PLA) hardly prints any plastic at all; in fact you can hear the brass head dragging over the glass PEI plate!
The right print head (Snapmaker White Support PLA) begins to print its initial pre-test strip; it for the most part prints normally.
RESULT: Failure with the X/Y test looking no different to METHOD 1 and the print head dragging across the print bed
METHOD 4

I have ran the 9 x 9 81 point bed-levelling calibration
Then ran the Z axis Sensor calibration. ← Because the Z Axis Auto Calibration doesn’t seem to be working.
Using the supplied white calibration card; I step the heads down manually until both Right and Left heads are touching the card, can be pulled toward me but cannot be pushed back under the head it wrinkles.
I then run the Z axis Auto Calibration
Then the X/Y axis calibration ← When running this calibration test the dual print head moves to the bottom left of the build plate
I manually update the Right and Left heads with an offset of anywhere 0.05mm all the way to 0.2mm
RESULT: Failure with the X/Y test. No combination of positive offset seems to satisfy this test; the patterns either print correctly on the far left and right sides in the X axis and not in the middle. The Y axis will print correctly the top half of the pattern and then not at all the bottom half. There doesn’t seem to be any happy medium (an offset) that is somewhere in the middle that results in a good print.
METHOD 5

Reset factory defaults – This wipes the newer firmware, and the device defaults the original firmware.
Follow the Out of the Box experience calibration method. I.e load filament → Level Bed → Calibrate.
Same result as METHOD 1
If I had to hazard a guess the Z Axis sensor that measures how far it is from the build plate is not functioning.

BTW at least two other owners of the Artisan have experienced the same issue; you can find their posts here. I tried using a 0.2mm Feeler Gauge with the manual Senor Calibration; this didn’t work for me.

Please advise on how to troubleshoot and fix this issue; as it stands my very expensive investment is completely unusable and isn’t making anything wonderful!

KarstenSchwab, 2023-03-30:
Hi . Same Problem by my artisan. Were is the 9x9 calibration find ? Thanks

ganymedes, 2023-04-10:
@mmack Any update on this case? Any success with the new test firmware SM3_V2.2.19_20230404.bin?

mmack, 2023-04-11:
@ganymedes finally someone who responds. Can you please identify who you are and for who you are speaking?

Printed half a bed and it did not scratch the surface anymore. However i changed something on the quick release myself to release pressure on the bed. I can't say what helped now. 
But what i can say is that i still have a gap between the heated bed and the print bed. Snapmaker should send us new beds which are even...

ganymedes, 2023-04-11:
@mmack Sorry for the confusion. I'm a customer like you are with the same issue. My first layer is ok on one side of the bed but scratching on the other. Bed levelling is not working correctly and it's very frustrating!

ganymedes, 2023-04-12:
@mmack ok, I might know what's going on. Auto levelling is pushing the nozzle to hard against the build plate. Therefore the build plate is bending a little bit and the reading is wrong. I did a whole 5x5 manual bed levelling and now it's working. Unfortunately the build plate is not heating when doing the manual procedure, so it's still not perfect but ok. Can someone test it as and confirm? Also I noticed that the build plate is only touching on four points. Is that on your Artisan also the case?

Alex0176, 2023-07-18:
i have exacly the same issue on my snapmaker 2.0 - i ordered the dual extruder module, because the leveling should be better, but its the same issue, since i get it from my kickstarter campane. 

[image]
[image]
[image]
[image]

r0dh3773, 2023-07-18:
Same issue here on a Snapmaker 2.0 A350, upgraded the linear modules and dual extrusion print head yesterday and cant get any usable bed leveling with the ABL / Auto Z-Offset.

The firmware was updated to the latest after the upgrade and the printhead is recognized correctly and everything appears to be ok.

But when it starts printing the calibration model it scratches into the build plate.

It doesnt look like the ABL mesh is being used at all…

WhiskeyBanger, 2023-07-27:
Im also having major issues with the leveling process. I spent all day trying to fix it with no end in sight. i did factory resets, tried everything. I hope snap maker fixes this issue.

BrennonTWilliams, 2023-07-28:
I'm having the same problem on my Snapmaker Artisan.

In this thread, users have provided a massive amount of data and documentation for Snapmaker devs to resolve this common and serious bug. Most Product Managers would kill for users that would give this much detail on bugs. If Snapmaker could muster half the enthusiasm and drive of its users, then bugs like this wouldn't be killing its brand.

This is why I no longer recommend Snapmaker to colleagues and friends. Users here serve up a wealth of thorough and thoughtfully collected measurements, log files, images, and samples. The Snapmaker team can't be bothered to even acknowledge they've looked.

Snapmaker, you don't deserve your users. Do better.

PEKuus, 2023-07-31:
Same problem here, tried auto, manual and sensor many time now. Nothing gave the correct results.

Zophiel, 2024-03-20:
This is exactly what is happening to my Artisan for a long time. I always thought it was due to my lack of knowledge.

- It would scrape at the edges of the build plate
- It would scrape across the print
- It would knock over the print at higher ranges.

This is causing VERY visible print lines. A new print module caused the same results. 
I think there is a discrepancy in the way it calculates the height and that accumulates the higher the print goes. Pressing into the 3D print.

Klabusterbaer, 2024-04-20:
Hello,

I have exactly the same problems. The automatic bed calibration doesn't really work. I also don't understand why Snapmaker can't manage to change the calibration from 350x350 to the 400x400 from the Artisan. It's as if you can only use 80m2 of a 100m2 apartment because the landlord only has 80m2 in the contract.

[image]

Here is an example after an automatic x/y and bed calibration.
[image]
[image]

- I can only get halfway reasonable pressure when I adjust the bed manually (9x9).

- In terms of feeling, I also think that the settings change on their own for no apparent reason. With one print everything goes well and with the next one with the same settings, the nozzle scrapes over the heated bed.

- Just now when printing a skirt, the nozzle scratched over the bed for a very short time, then swung to the side and then moved back to its correct position and continued printing.
[image]

- It also happens again and again that the nozzle suddenly moves through the middle of the print despite the z-jump.
[image]

- I also find the Snapmaker support video on YouTube about calibrating the sensor strange. It is not set there as described on the screen, but rather the calibration card must be able to be easily pulled out and pushed forward (without a fold).

Here is the YouTube link from the Snapmaker support video about sensor calibration:
[https://youtu.be/gqDGZTLWSPo?list=PLZiH-tfTScQJDgedKyLJDyj_SjvoTjl71](url)

ps: sorry for my bad google englisch :-)

scotthsl, 2024-11-15:
Hello Mack and everyone who shared feedback on this issue, @mmack @ganymedes @Zophiel @Klabusterbaer @PEKuus @BrennonTWilliams @WhiskeyBanger @Th199403 @jakubkinst 
First, we want to thank you for your patience and detailed feedback. We recognize that our response here has been delayed, and we apologize for not addressing this issue sooner on GitHub. We did actively follow up on this issue through our support tickets, and since then, we have worked on specific firmware updates to address the auto-leveling challenges described here.

Issue Summary and Fixes
We found two key problems with the previous firmware. Here’s a breakdown of the solution:
1. Live Z Offset Issue & Fix
We identified a firmware bug that occasionally caused the Z-axis to compensate in the opposite direction when adjusting the live Z offset. For example, attempting to raise the nozzle sometimes resulted in downward pressure. This issue was fixed in Artisan Firmware v2.2.3 (released on March 31, 2023).
2. Segmented Compensation Optimization
During testing, we found a bug in segmented compensation functionality. With this bug resolved, segmented compensation now functions as expected. When the firmware receives a movement command in the XY plane that crosses multiple leveling grid points, the movement is now divided according to grid boundaries, applying multiple Z-axis compensation points throughout. This improvement provides better compensation during movement across the bed, helping to prevent inconsistent extrusion along a single line (over-tightening or loosening issues). This fix assumes that leveling data is accurate, without significant platform deformation. These updates were included in Artisan Firmware v2.2.4 (released on April 14, 2023).
Access all Artisan firmware downloads and updates here: https://forum.snapmaker.com/t/snapmaker-artisan-firmware-updates-and-downloads/29975

Further Troubleshooting Guidance
For anyone who continues to experience bed leveling issues even after updating to Firmware v2.2.4 (released on April 14, 2023) or later, we recommend checking the gap between the support platform and printing platform on your Artisan machine. In some cases, adding one or two shims (0.5–0.8 mm thick) between these two platforms would help resolve persistent leveling issues due to platform flatness. You can find additional troubleshooting steps here: [Snapmaker Troubleshooting Guide on First Layer Adhesion](https://wiki.snapmaker.com/en/general/troubleshooting/poor_first_layer_adhesion#the-support-platform-is-uneven-only-for-artisan).

We value the community's input, which has greatly assisted us in resolving these issues. If you have questions or continue to have bed leveling problems, contact the Snapmaker support team by submitting a ticket at https://snapmaker.formcrafts.com/support-ticket. The support team typically responds within two business days. Please include as much information and as many photos as possible in your submission to expedite the process.

Best regards,
The Snapmaker Teamg

closed 2024-11-15 by scotthsl