Snapmaker/Snapmaker-Artisan-Firmware-Issues #14: Bug: Allways 100% speed shown on Artisan display

Obicom, 2023-05-31:
Luban Version 4.8.0 (and before)
Is this a regression? (optional)
Yes

To Reproduce:
Steps to reproduce the behavior:
Start a laser job and take a look on the Artisan 7'' display.

The power gauge is shown the correct value in the middle of the gauge.
The speed gauge shows allways 100% as a figure in the middle of the gauge. - WHY???

🌍 Your Environment
Platform:
Windows 11
Luban 4.8.0 release version (latested)
Snapmaker Artisan with Rotary module and 10W laser

[image]

matze19999, 2023-06-26:
This is not a bug, this works as intended. Workspeed 100% = what you set in Luban as workspeed, for example 2.500mm/m.

You can set it to 200% on the display, which would be 5.000mm/m then.

I dont know why you think this is a bug.

Obicom, 2023-06-26:
Easy to explain, independent of my settings it shows always 100% and the gauge should show the value also ... so is it 100% it should be also a full circle of the gauge. I would also expect that the value (100%) reflects the maximum of speed ... otherwise is really confusing.

matze19999, 2023-06-26:
Absolutely not, because the 100% is based on the speed you have set in luban for this laser job, not the maximum what the printer can do.

100% = Max printer speed wouldn't make sense because you could not set the temporary speed exactly higher or lower based on the actual speed. 

Sorry, but this whole Issue can be closed.

Obicom, 2023-06-26:
@matze19999 I would disagree to close the topic. Why it shows then to different values? The gauge and the number should be the same. It is total confusion that the number should reflect the "relative" speed from the configured one and the guge reflects the total  maximum speed. If so as you explained, minimum both should show the same value like the laser power.